xioTechnologies/Gait-Tracking-With-x-IMU #4: What type of binary files are being read in the datasets

kswain1, 2016-07-14:
Currently have my data in csv format, but I don't understand how to convert it into .BIN format for it to be read by the matlab program. Is there any way around this issue?

ybcho, 2016-09-08:
I really hope to know how I can change my own .CSV file to .BIN file...plz help me...

debajyoti93, 2017-04-27:
Facing the same issue. Please help me

VagabundoPL, 2017-05-10:
make sure, that your initPeriod value is not bigger than the last value of paket number in input file.

ozgetuncer, 2018-02-27:
I have also have the .csv file and need to .bin file. How did you able to read the data?

xioTechnologies, 2018-02-27:
The x-IMU logs data to the SD card as a .bin file.  The .bin file can then be converted to CSV files using the x-IMU GUI.  The .bin is not used by the MATLAB script.

closed 2018-02-27 by xioTechnologies

MuhammadAli152, 2021-01-02:
I am getting the following error.

Index exceeds matrix dimensions.

Error in Script (line 109)
indexSel = 1 : find(sign(time-(time(1)+initPeriod))+1, 1);

Please Help me in this regard Thank You.

Wangkingling, 2021-09-22:
> 我犯了以下错误。
> 
> 索引超过矩阵维度。
> 
> 脚本错误（行 109）
> 索引塞尔 = 1 ： 查找 （符号 （时间 -（1）=initPeriod）=1，1）：
> 
> 请在这方面帮助我谢谢。

    % Compute integral drift during non-stationary periods
    velDrift = zeros(size(vel));
    stationaryStart = find([0; diff(stationary)] == -1);
    stationaryEnd = find([0; diff(stationary)] == 1);
    if length(stationaryStart) ~= length(stationaryEnd)
        for i = 1:numel(stationaryEnd)-1
            driftRate = vel(stationaryEnd(i)-1, :) / (stationaryEnd(i) - stationaryStart(i));
            enum = 1:(stationaryEnd(i) - stationaryStart(i));
            drift = [enum'*driftRate(1) enum'*driftRate(2) enum'*driftRate(3)];
            velDrift(stationaryStart(i):stationaryEnd(i)-1, :) = drift;
        end
    else
        for i = 1:numel(stationaryEnd)
            driftRate = vel(stationaryEnd(i)-1, :) / (stationaryEnd(i) - stationaryStart(i));
            enum = 1:(stationaryEnd(i) - stationaryStart(i));
            drift = [enum'*driftRate(1) enum'*driftRate(2) enum'*driftRate(3)];
            velDrift(stationaryStart(i):stationaryEnd(i)-1, :) = drift;
        end
    end