xioTechnologies/Gait-Tracking-With-x-IMU #5: How to adjust the sampling time? (2sec and  1sec)

AlexWUrobot, 2016-07-17:
Because I already set "Data Output Setting" frequency from 1Hz to 512Hz
but I still can not get the sample data like below
Packet number	Gyroscope X (deg/s)	Gyroscope Y (deg/s)	Gyroscope Z (deg/s)
3	-0.0625	-0.25	-0.0625
5	-0.0625	-0.25	-0.0625
6	-0.0625	0.0625	0
8	-0.375	-0.25	-0.375
9	-0.0625	0.5625	0.0625
11	-0.125	-0.125	-0.1875
12	0.125	0.5	-0.0625
I mean my Packet number can not variable intervals
5-3=2
6-5=1
8-6=2
9-8=1
Most of all, I can not run the program 
Matlab ERROR like below
 
Error in ==> Script at 111
indexSel = 2 :
find(sign(time-(time(1)+initPeriod))+1,1);

demo-wang, 2017-03-10:
I think you should read Script code carefully, in the demo the sample time is 1/256, that mean in 1 s, there will be 256 packet received; in you demo you just change samplePeriod to 512, and do not forget change the value of startTime(s) and stopTime(s), the period time you want to use.

ahmednawazkhan, 2019-02-21:
Thanku @OceanWong1991 . Can u please explain start time and stop time in this case?

xioTechnologies, 2019-02-28:
startTime and stopTime are used to frame the data so that only the period of interest is processed

ahmednawazkhan, 2019-03-04:
can u explain a litter more please

Omegaki113r, 2019-03-10:
Hi 

I worked with this and had the same issue 

I am new to matlab and programming but i think my interpretation is correct.

When you import your dataset according to the sample rate the time for each readings of data set is given by the script.

Assume like this you have 10000 readings in your csv file if your sample rate is 1/256 the first reading is given 1sec and your 10000th reading will given 256sec ( just assumptions)

Then in the script there is a start time and end time variables assume that 
Start time = 10 
End time = 50

Then the script only use the readings between 10 and 50 seconds 0-10 and 50-256 sec are ignored 

Did you understand now.

If not I can explain more.

ahmednawazkhan, 2019-03-10:
what if i want to use sample rate = 1. what does this mean and what will be the start and end time in this case? suppose i have 1000 samples

Omegaki113r, 2019-03-10:
As I told you I am new to this as you.

But let me get back to you in a while.

I will answer soon

ztrattner, 2019-03-18:
Despite following the advice of OceanWong and attempting to follow along with Omegaki I am receiving the same error as the above posters. I am currently working with these key assumptions:
1. I have checked that the input units being used are the same (°/s for gyro, g for accelerometer, G for magnetometer).
2. My sample period is 1/256, implying 256 samples per second. To me this means that packet numbers 1-256 will belong to second=1, 257-512 to second=2, etc.
With 2560 packets, I would expect to have my stop time go up to 10, but if I put any number higher than 1 I get the error:

Index exceeds the number of array elements (0).
Error in Script (line 113)
indexSel = 1 : find(sign(time-(time(1)+initPeriod))+1, 1);

Is there anybody with a solution to the above error?

ahmednawazkhan, 2019-03-30:
what is he trying to do with "indexSel". what is it

IonRpd, 2019-06-06:
> Despite following the advice of OceanWong and attempting to follow along with Omegaki I am receiving the same error as the above posters. I am currently working with these key assumptions:
> 
> 1. I have checked that the input units being used are the same (°/s for gyro, g for accelerometer, G for magnetometer).
> 2. My sample period is 1/256, implying 256 samples per second. To me this means that packet numbers 1-256 will belong to second=1, 257-512 to second=2, etc.
>    With 2560 packets, I would expect to have my stop time go up to 10, but if I put any number higher than 1 I get the error:
> 
> Index exceeds the number of array elements (0).
> Error in Script (line 113)
> indexSel = 1 : find(sign(time-(time(1)+initPeriod))+1, 1);
> 
> Is there anybody with a solution to the above error?
> 
> EDIT: I cannot empirically verify the 2nd assumption above since my sensor only goes up to 100 Hz (MPU9250 running at 500000 baud through Arduino Uno). Kind of looks like an error in the code but my matlab experience is limited so can't be sure.

In Script.m put a break point  right after time variable, then open the time variable. It should be populate with all moments of time you write in file according of  your samplePeriod. So for example first value is 0(first written value) and the last value is 9(last value saved was 9 second later), you can't set stopTime grater then 9 second. This is the cause of 'indexSel' error;

MuhammadAli152, 2021-01-02:
I am also getting the following error.

Index exceeds matrix dimensions.

Error in Script (line 109)
indexSel = 1 : find(sign(time-(time(1)+initPeriod))+1, 1);

Please Help me in this regard Thank You.