xioTechnologies/Gait-Tracking-With-x-IMU #6: error: No data was imported

ozanAta, 2017-04-04:
Hi,
I am new to MATLAB/Octave and currently when i run the Script file on Octave i get the following error

`
error: No data was imported. 
error: called from
    xIMUdataClass at line 49 column 17
    Script at line 26 column 10
`

weibo053, 2017-12-11:
This happens to me as well, please help

ozgetuncer, 2018-03-02:
I get this error too and need help.

Error using xIMUdataClass (line 49)
No data was imported.

Error in Script (line 26)
xIMUdata = xIMUdataClass(filePath, 'InertialMagneticSampleRate', 1/samplePeriod);

xioTechnologies, 2018-03-02:
I have just downloaded the repository from GitHub and run the script in MATLAB R2017a without issue. 
 "No data was imported." is an error message created by the x-IMU MATLAB Library.  I suggest you investigate what the source error is.

closed 2018-03-02 by xioTechnologies

daylanKifky, 2019-12-16:
Trying to run it with `octave` does raise the described error.

the error inside the `xIMUdataClass` is:

```
parse error near line 27 of file /home/daylan/X/repos/Gait-Tracking-With-x-IMU/Gait Tracking With x-IMU/ximu_matlab_library/TimeSeriesDataBaseClass.m

  external methods are only allowed in @-folders

>>>         Plot(obj);