xioTechnologies/Gait-Tracking-With-x-IMU #7: Reading in my own CSV file of data

CarbontTitan, 2017-08-31:
I have collected my own data in CSV format, however, how do I get it read into the program?
How do you create the BIN files?
Many thanks.

xioTechnologies, 2017-09-11:
The BIN file is not used by the MATLAB script.  It is only included for reference.

syahrulizmeer, 2017-12-26:
then how we want to add our own data?
Much appreciated for the reply
thank you

declanxyz, 2018-01-14:
seems that we are meeting the same problem.
what we want to know the most is how to run the code using our own data?
really appreciate it if this problem can be solved.

xioTechnologies, 2018-01-16:
The code is provided with example data and runs without issues.  You can modify the code or the data however you like.  There are no tutorials for this.

closed 2018-01-16 by xioTechnologies