xioTechnologies/Gait-Tracking-With-x-IMU #9: Undefined function or variable 'xIMUdataClass'.

declanxyz, 2017-11-28:
Firstly, thank you for your code. I'm new to matlab. When I run Script I get the following error:

Undefined function or variable 'xIMUdataClass'.

Error in Script (line 26)
xIMUdata = xIMUdataClass(filePath, 'InertialMagneticSampleRate',
1/samplePeriod);

vesper305, 2017-11-28:
Hope this email finds you well.

I think you didn’t add the file ‘xIMUdataClass.m’ into the system path. This is a self-defined handle use to abstract data from the data files.
Actually, I think you should add the whole project into the system path, otherwise you will get more error like this.

If you have any problem, please don’t hesitate to tell me.

Best,
Guo Liu

declanxyz, 2017-12-22:
Thank you so much for your kindness and your help. I have solved the problem by adding the whole project into the path, but now I have met another problem. I hope I can get more help from you. I built a new bin file and tried to read my own data in a csv file, the program returns "No data was imported." and I get the following error:

error in Script (line 30)
xIMUdata = xIMUdataClass(filePath, 'InertialMagneticSampleRate', 1/samplePeriod);

Looking forward to your reply, and again, thank you so much.