xioTechnologies/Gait-Tracking-With-x-IMU #11: About the time data

ZM-NEU, 2018-03-13:
First thanks a lot for the code.But when I tried to use the code with my own IMU data I ran into some problems: data from my csv file can be imported but the results came out to be null arrays, and I thought it might have something to do with time array. The time array is used in the program but I haven't found any time data in the example csv files provided, and I'm a bit confused where the time array came from, are they simply generated from the startTime and stopTime or what?
Much appreciated for the reply!

yekanchi, 2018-03-14:
What Sensor you Use?

ZM-NEU, 2018-03-15:
I use a x-sens IMU sensor

yekanchi, 2018-03-17:
Check the Gyro Output units, 
check it if it's in deg/s or it is in rad/s, i had the same issue

ZM-NEU, 2018-03-17:
Thanks a lot for your advice! I did ignore that and now I have corrected the data to deg/s, but it still doesn't work. How did you fix your issue? Is there anything else to keep in mind?

yekanchi, 2018-03-17:
there may be other issues too, for instance the Sample Rate is crucial too. you can send me your sample data if you want and i can review it if you mind, of course it will take within next two weeks,

ZM-NEU, 2018-03-19:
Thanks a lot!! Here is my data, and please inform me if there's any problem with it. The data include a csv file and the original txt data converted from the rosbag file the sensor generated. Much appreciated for your kind help!
[myIMUdata.zip](https://github.com/xioTechnologies/Gait-Tracking-With-x-IMU/files/1823431/myIMUdata.zip)

Momo7617, 2018-03-26:
Hello guys, thank you very much for the codes, I am working on a project where I have to measure the length of the step for gait analysis, so I need to measure the position from the acceleration using (BNO55 IMUs) .....
I want to ask you which modifications must be imported to the matlab code or C# code in order to do my project ??? 
Thanks in advance

ZM-NEU, 2018-03-30:
I suggest you open another issue about your problem

ZM-NEU, 2018-04-16:
I recorded another data set with magnetic data, at sample rate of 100Hz this time, but it still didn't work. I'd really appreciate it if anyone can view my data and figure out what's the problem with it
[MyData-sampleRate100.zip](https://github.com/xioTechnologies/Gait-Tracking-With-x-IMU/files/1915116/MyData-sampleRate100.zip)

ZM-NEU, 2018-04-16:
By the way, there's another problem I am confused about: How come the packet number of the example data discontinuous? Since I have problem generating the right time vector, I wonder if there's something to do with my packet number, but my packet number is almost continuous while it seems to be lots of packets missing in the example data. For example there's 15154 packets in stairsAndCorridor dataset, but the packet number reached 23026. Is there anything to notice about packet numbers? Thanks in advance!

ZM-NEU, 2018-04-18:
OK，this problem is fixed, it's because I misunderstood the meaning of startTime and stopTime.

closed 2018-04-18 by ZM-NEU

mceuint, 2018-12-01:
Could you explain what startTime and stopTime is? I am having the same issue.

TAREMC, 2018-12-01:
@ZM-NEU, could you explain what you found the meaning of startTime and stopTime to be? I'm running into the same issue. Thanks in advance!

ZM-NEU, 2018-12-02:
@TAREMC ，it's simple: startTime and stopTime help you pick up the part of data you want to use from your dataset. startTime is where the data you want to process and visualize starts while stopTime is where it ends, you can change them to choose different part of data in your dataset. They are measured in seconds, computed from divide the number of data packets by your sample rate, so also make sure to set the sample rate to that you used recording the dataset.

ZM-NEU, 2018-12-02:
@TAREMC ，it's simple: startTime and stopTime help you pick up the part of data you want to use from your dataset. startTime is where the data you want to process and visualize starts while stopTime is where it ends, you can change them to choose different part of data in your dataset. They are measured in seconds, computed by dividing the number of data packets by your sample rate, so also make sure to set the sample rate to that you used recording the dataset.