xioTechnologies/Gait-Tracking-With-x-IMU #12: Errors when taking turns

Oshel, 2018-03-28:
Hello,

First of all- thank you for making the code public. I would like to ask for a help or a hint.

I've edited your code to take samples from my custom file. I provide the program with time, clean accelerometer data and gyro data with offsets subtracted. I use LSM9DS0 sensor with 190Hz (gyro) / 200Hz (acc). Straight line walking works well. But when I try to take a turn, then I gain the position Z axis offset and the the trajectory of the returning path is not parallel to the original. Photos attached shows it best (red path is the actual path). I am wondering where is the problem? It occures that the AHRS algorithm does not work well with my data. Any ideas?

Thank you in advance :)

[image]
[image]

Oshel, 2018-03-29:
Hello again,

I managed to apply your IMU filter (accelerometer and gyro) directly on MCU. It occures that with my data the yaw axis is floating. Pitch and roll stands still (stationary position on my desk). After that I tried your MARG filter with calibrated magnetometer- the yaw is not floating anymore.

Is there a way to adapt MARG filter in the gait tracking algorithm? I can see that you use
`UpdateIMU(obj, Gyroscope, Accelerometer)`
function. The function
`Update(obj, Gyroscope, Accelerometer, Magnetometer)`
is "not implemented".

xioTechnologies, 2018-04-16:
Drift is always expected when using numerical integration methods.  The error may be reduced by improving sensor calibration or adjusting the algorithm behavior; e.g. when the identified stationary period starts/stops.  The details of this are too complicated to elaborate on here.

closed 2018-04-16 by xioTechnologies