xioTechnologies/Gait-Tracking-With-x-IMU #13: IMU data drifting

wippp, 2018-04-05:
Hi!

I was very impressed with your work with the Gait tracking algorithm and wanted to test it with my MPU-9250 sensor. I strapped the sensor onto my ankle and walked on a treadmill while sampling data. I then managed to feed my data into your Matlab script and got a result but the output drifts several meters.
I will include a picture of the output and perhaps someone could help me figure out what I am doing wrong...

Best Regards
Anders Persson
[image]

xioTechnologies, 2018-04-16:
The IMU must be attached to the foot, not the ankle.  A sensor attached to the ankle will never be stationary; this is a fundamental requirement of the algorithm.  The algorithm will not work with a treadmill for the same reason.

closed 2018-04-16 by xioTechnologies

wippp, 2018-04-16:
Thank you for answer! I realised that also after a while.... Did some test with an IMU attached to the foot resiliently with much better results.