xioTechnologies/Gait-Tracking-With-x-IMU #19: How to add your own CSV file data ?

NalinLuthra, 2018-09-29:
I want to visualise my own data IMU data how to change from .csv to .bin to allow matlab to read it ?

xioTechnologies, 2019-01-11:
The .bin file is not used.  The .bin file was recorded using the x-IMU and then converted to the CSV files.

zhangshengkai520, 2019-01-11:
Hi, I also want to use my own .csv DATA, but the Matlab give me the error. 
Error using xIMUdataClass (line 49)
No data was imported.
Error in Script (line 26)
xIMUdata = xIMUdataClass(filePath, 'InertialMagneticSampleRate', 1/samplePeriod);

Could you explain it more detial？

SongZhuhai, 2019-01-12:
你们自己的csv数据格式和她提供的格式不一样吧，而且精确度和单位都要注意一下

zhangshengkai520, 2019-01-12:
Did you try these matlab codes?  Could you leave you contect information to me?  @songzhuhai 
 And also, how to transfer my own data to the csv file?

SongZhuhai, 2019-01-12:
MATLAB我不太清楚，当初用Python写的，这个MATLAB开源版也被我改成了Python版本。

SongZhuhai, 2019-01-12:
如果使用这个开源源码运行，你们可以参照https://github.com/xioTechnologies/Gait-Tracking-With-x-IMU/blob/master/Gait%20Tracking%20With%20x-IMU/Datasets/straightLine_CalInertialAndMag.csv这个格式改数据，注意存放目录和计量单位就可以了。

zhangshengkai520, 2019-01-12:
@songzhuhai 大神， 我也在用python做， 但是始终位移什么的始终对不上。 你用的是什么imu， 如果可以的话，能否参考一下你的python代码。我 一直不出结果，很着急。。或者我真的有很多问题想要请教你。。能否加一个微信或者别的联系方式。。我已经做了很久了。

zhangshengkai520, 2019-01-12:
@songzhuhai, 我是按照这个csv 文件改的数据， tracking的那个可以读取， 但是这个不行， 提说都不到gyro的数据；。。我现在头都大了。真的迫切需要和你联系一下。。请教问题

Omegaki113r, 2019-03-07:
Did anyone found a solution for this?

I am using a custom setup. MPU6050 and a HMC5883L for acquiring data suing an Arduino nano to process and save the data in a SD card module. 
when i tried to use the csv file created by my system. importing fails. this is the format of my data.

Packet number,Gyroscope X (deg/s),Gyroscope Y (deg/s),Gyroscope Z (deg/s),Accelerometer X (g),Accelerometer Y (g),Accelerometer Z (g),Magnetometer X (G),Magnetometer Y (G),Magnetometer Z (G)
404,0,0,0,2.4,1.63,-10.69,0,442,44
1296,0,0,0,2.36,1.64,-10.59,-4,448,40
1380,0,0,0,2.35,1.62,-10.67,-4,446,42
1469,0,0,0,2.3,1.58,-10.44,-6,441,37
1558,0,0,0,2.35,1.73,-10.59,-3,446,36
1648,0,0,0,2.36,1.65,-10.59,-1,443,36
1737,0,0,0,2.36,1.61,-10.58,-2,447,44
1827,0,0,0,2.42,1.69,-10.59,-1,444,38
1916,0,0,0,2.36,1.69,-10.63,-7,450,40
2006,0,0,0,2.36,1.65,-10.62,-6,442,36
2095,0,0,0,2.38,1.62,-10.59,-6,446,38
2172,0,0,0,2.37,1.59,-10.62,0,443,38
2273,0,0,0,2.36,1.66,-10.61,0,445,35
2361,0,0,0,2.37,1.67,-10.6,-2,442,36
2451,0,0,0,2.44,1.64,-10.58,-2,447,38
2540,0,0,0,2.39,1.66,-10.67,-5,449,40
2629,0,0,0,2.38,1.63,-10.58,-2,446,39
2719,0,0,0,2.4,1.66,-10.57,-5,445,42
2808,0,0,0,2.37,1.64,-10.55,-2,443,36
2898,0,0,0,2.37,1.67,-10.59,-6,445,40
2988,0,0,0,2.38,1.64,-10.53,0,439,37
3076,0,0,0,2.36,1.7,-10.57,-5,445,37
3171,0,0,0,2.34,1.64,-10.57,-4,442,38
3253,0,0,0,2.34,1.7,-10.63,-3,448,40
3343,0,0,0,2.38,1.63,-10.6,-5,445,40
3432,0,0,0,2.33,1.66,-10.55,-7,447,38
3521,0,0,0,2.37,1.67,-10.62,-3,444,41
3611,0,0,0,2.35,1.62,-10.56,-2,445,35

Omegaki113r, 2019-03-10:
I found how to add our own files to the matlab code import them properly and plot the graphs.

Anyone need?

ahmednawazkhan, 2019-03-10:
yes. can u please share

pshah93dbz, 2019-03-15:
> I found how to add our own files to the matlab code import them properly and plot the graphs.
> 
> Anyone need?

@Omegaki113r please can you share

bimodimas, 2019-03-27:
> 
> 
> I found how to add our own files to the matlab code import them properly and plot the graphs.
> 
> Anyone need?

@Omegaki113r yes please, i need that

Omegaki113r, 2019-03-27:
> > I found how to add our own files to the matlab code import them properly and plot the graphs.
> > Anyone need?
> 
> @Omegaki113r yes please, i need that

Hi,

assume that you need to use strainght line walking data set plotting. look at the example data file name...

straightLine_CalInertialAndMag.csv

so if you wanted to add your own data set use,

xxxxx_CalInertialAndMag.csv

as the name..

and

in the Script.m line 10 change the filePath variable as below,

filePath = 'Datasets/xxxxx';

ningquewulan, 2019-04-21:
我将IMU与串口连接，想要将程序改成实时绘图的，是不是只要将它的accXYZ,和gryXYZ，改成自己的就行了，出现的一个问题是能绘出图来，但是在调用SixDofAnimation这个文件的时候，每次图都是从头开始绘制

MuhammadAli152, 2021-01-02:
I am getting the following error.

**Index exceeds matrix dimensions.

Error in Script (line 109)
indexSel = 1 : find(sign(time-(time(1)+initPeriod))+1, 1);**

Please Help me in this regard Thank You.

Kamlesh364, 2021-07-10:
> I am getting the following error.
> 
> **Index exceeds matrix dimensions.
> 
> Error in Script (line 109)
> indexSel = 1 : find(sign(time-(time(1)+initPeriod))+1, 1);**
> 
> Please Help me in this regard Thank You.

I'm also getting the same!