xioTechnologies/Gait-Tracking-With-x-IMU #21: Can we use this for tracking a UAV trajectory using the flight controllers IMU?

ahmednawazkhan, 2019-02-21:
Hi. thank you for the code. It is extremely use full for learning. I was wondering if I can use this code to estimate my UAV flight trajectory using same IMU readings from the Flight controller of UAV.

xioTechnologies, 2019-02-28:
The gait tracking code is for tracking the motion of a foot during walking.  The code cannot be used to track a UAV.

ahmednawazkhan, 2019-03-04:
and why is that sir?

rolba, 2019-03-04:
In my opinion it's possible. Gait-Tracking-With-x-IMU is nothing else than inertial navigation system. Probably you can use this code to estimate the relative position from your current position.

ahmednawazkhan, 2019-03-04:
This is what I believe too. but inside the code ````script.m```` the data is split into chunks. may be it is kind of taking average or something for accuracy and hence requires a large amount of data. how can I deal with that?

rolba, 2019-03-04:
Where exactly do you see that?

xioTechnologies, 2019-03-04:
The tracking relies on the sensor being stationary at the end of each foot stride, e.g. every 1 second.  These frequent stationary periods are essential to compensate for drift the in calculated velocity.

ahmednawazkhan, 2019-03-04:
surely i would like to ignore this drift error correction and try to approximate the trajectory from this since i have found no other free software that enable me produce my UAV flight trajectory. So i think I can ignore this drift error correction?

ahmednawazkhan, 2019-03-04:
@rolba here is the part which seems to split data 

```
% Initial convergence
initPeriod = 2;
indexSel = 1 : find(sign(time-(time(1)+initPeriod))+1, 1);
for i = 1:2000
    AHRSalgorithm.UpdateIMU([0 0 0], [mean(accX(indexSel)) mean(accY(indexSel)) mean(accZ(indexSel))]);
end
```

i am not getting this part at all :)

```
% For all data
for t = 1:length(time)
    if(stationary(t))
        AHRSalgorithm.Kp = 0.5;
    else
        AHRSalgorithm.Kp = 0;
    end
    AHRSalgorithm.UpdateIMU(deg2rad([gyrX(t) gyrY(t) gyrZ(t)]), [accX(t) accY(t) accZ(t)]);
    quat(t,:) = AHRSalgorithm.Quaternion;
end
```