xioTechnologies/Gait-Tracking-With-x-IMU #27: LP filter accelerometer data help

arcgislearner, 2019-10-30:
Hello, I want to know how to determine the value of the filtCutOff in Script.m：

filtCutOff =5;
[b, a] = butter(1, (2*filtCutOff)/(1/samplePeriod), 'low');
acc_magFilt = filtfilt(b, a, acc_magFilt);

MTfirst, 2019-12-09:
In my experience, about 1 Hz to 6 Hz is brought by subjects.
In other words, f < about 1 or f > about 6 is noise (for example: offset of IMU) so we can cut.

Though filtCutOff=5 is valid of most subjects, if you want define manually, Upper Band should determined according to walking speed.

Thanks.

arcgislearner, 2019-12-10:
OK,thanks for your reply, i will try it again.

 In other words, f < about 1 or f &gt; about 6 is noise (for example: offset of IMU) so we can cut.

Abdul-Sajeed, 2020-10-09:
Hello, what if i am using vehicle with speed of 20 - 30 km/hr. Is it possible to reset cutoff speed ? (thanks)

arcgislearner, 2020-10-12:
Sorry, I didn't do this in the car, I just tied it to my feet.