xioTechnologies/Gait-Tracking-With-x-IMU #32: Navigation frame help ?

Abdul-Sajeed, 2020-10-06:
Hello, i have been using accelerometer and gyroscope data to determine and track position of ground vehicle. But when i see script position data figure is some what confusing. I would like to know what navigation frame is been used for position ? thanks.

xioTechnologies, 2020-10-06:
The code uses NWU.  X= North, Y = West, Z = Up.

Abdul-Sajeed, 2020-10-09:
Thank you for the reply, 
In your case, For straight line, stairs and corridor, and for spiral stairs. Translation position in 'm' was close to real distance ?

Abdul-Sajeed, 2020-10-09:
Thank you for the reply,
Is the frame fixed  ? Where is the origin of the frame ?
and In your case, For straight line, stairs and corridor, and for spiral stairs. Translation position in 'm' was close to real distance ?

xioTechnologies, 2020-10-13:
The origin is the starting position.  The accuracy for the walks was not evaluated.

Sinou12, 2022-06-22:
First, thank you, @xioTechnologies,  for the code and your feedback.
I'd like to get an answer if it's possible!
As I understand it, the frame of reference is the NWU land frame. And you chose the origin as the starting position. How did you choose it? Is it in the code? If yes, in which part of the code? 
Thank you for your collaboration

Sinou12, 2022-06-22:
Normally, there is the sensor frame and the reference frame. So to calculate the position in the reference frame, we must transform the values of the acceleration and the angles of rotation to calculate the quaternion. 
So according to your answer, the reference frame moves every time the foot touches the ground and it is initialized to X=0, Y=0 and Z=0. And calculate the position of this frame?
The diagram in attachment for more clarification.
  
Thank you for your feedback.

[image]

xioTechnologies, 2022-06-22:
I am unable to follow your reasoning.  I suggest that you run the code with the example data, perhaps that will clear things up for you.  I also suggest that you instead use the [Python version](https://github.com/xioTechnologies/Gait-Tracking) as the code may be easier to follow.

Sinou12, 2022-06-22:
All right, thank you.