xioTechnologies/Gait-Tracking-With-x-IMU #34: Problems about using my own IMU in this project

tianyazhe233, 2020-11-14:
I have tried this project with my own IMU, the sample frequency is 200HZ, but the result is really very different from the ground truth. I just modified the samplePeriod in the Scrip.m file, so I want to know if there any other parameters that I need to change, beside, in this project and another project 'Oscillatory-Motion-Tracking-With-x-IMU-master', you both used HP filter and Low filter for processing, can you explain why you do it in this way? Very thanks.

xioTechnologies, 2020-11-15:
The high-pass filter compensates for the drift that results from numerical integration.  This repository provides source code and example data so that others can play with the code.  It is not a tutorial and will be of little use to anyone who wishes to it as a tool without understanding the code.

The sample rate (line 25) must match the logged data.  The start time and stop time (lines 11 and 12) crop the data to the period of interest.  All other parameters should be understood from the comments and by playing with the code.

tianyazhe233, 2020-11-16:
> The high-pass filter compensates for the drift that results from numerical integration. This repository provides source code and example data so that others can play with the code. It is not a [...]

Very thanks, when i try to running the code section by section,  trying to understand what happend during each stage, I'm confused about this 'for loop', it seems the variable i has no effect in this loop, so is there something misssed here?

% Initial convergence
initPeriod = 2;
indexSel = 1 : find(sign(time-(time(1)+initPeriod))+1, 1);
for i = 1:2000
    AHRSalgorithm.UpdateIMU([0 0 0], [mean(accX(indexSel)) mean(accY(indexSel)) mean(accZ(indexSel))]);
end

xioTechnologies, 2020-11-16:
That section of the code is titled "Initial convergence".  The AHRS algorithm update method is called 2000 times with the same accelerometer measurement (calculated as mean of first 2 seconds) so that the AHRS algorithm converges to this initial inclination.  The "i" is not used in the loop it is only used to create the loop.