xioTechnologies/Gait-Tracking-With-x-IMU #35: Use MPU9250 to walk a straight line, the Y-axis data of Position will deviate to the left or right

ShihEn, 2020-12-15:
Thank you for sharing the code

But I want to ask you some questions about the usage. Walking along a straight line, the Y-axis will be offset from the results of Position data analysis. Is this due to the offset caused by walking towards the magnetic north or the gyroscope when walking?
My data results
[image]
[image]

pajaraca, 2021-01-05:
Hi! I believe this is just integration errors. Your device is a low cost IMU with lot of noise, and the double integration in this code isn't powerfull enough to deal with it (even tough the results are awesome).
PS - this code do nothing with mag measurements, just accel and gyro are needed.

celinlab, 2022-04-26:
Hi, 

I have the same issue. Do you resolve it ? 
[image]