xioTechnologies/Gait-Tracking-With-x-IMU #36: Can this code use for approximate phone tracking?

BangwenHe, 2021-07-10:
Hi, thanks for sharing this awesome code. I want to use sensors in smart phones to track the 3D position of the phones. I tried some ways to track before I saw your result in Youtube. 

I change my output to fit your input. But I didn't get a convincing result from this code. I see that you said stationary period during the whole process is essential to compensate the drift. So I want to ask you whether it's possible to track the 3D trajectory of a smart phone, or any possible way I can use to do it. Any advice or comment will be appreciate.

xioTechnologies, 2021-07-13:
The gait tracking code is for tracking the motion of a foot during walking.  The motion of a smart phone is very difficult and so this code will not work.

BangwenHe, 2021-07-14:
> The gait tracking code is for tracking the motion of a foot during walking. The motion of a smart phone is very difficult and so this code will not work.

Thanks for your reply. We have changed our goal to find the time gap between the maxima and minima in the sensor outputs.

closed 2021-07-15 by BangwenHe