xioTechnologies/Gait-Tracking-With-x-IMU #37: Error Inputting Data

austin51, 2021-07-15:
I am attempting to input my own data, with columns:

Time (sec) | ARS1 (deg/s) | ARS2 (deg/s) | ARS3 (deg/s) | Acc1 (g) | Acc2 (g) | Acc3 (g)

Is there a way to do this properly? I am getting error:

Error using xIMUdataClass (line 49)
No data was imported.

Error in Script (line 32)
xIMUdata = xIMUdataClass(filePath, 'InertialMagneticSampleRate', 1/samplePeriod);

Thank you!

xioTechnologies, 2021-07-19:
The CSV file format needs to be the same as the example data.  The column headings are ignored by the code, they only exist for a human reader.

exilat, 2022-05-16:
Hi ,
I also want to input my own data, i did followed the same format albeit i don't have magnetometer data, so i filled them with 0.
I am getting:
 Error using xIMUdataClass (line 49)
No data was imported.
Error in Script (line 26)
xIMUdata = xIMUdataClass(filePath, 'InertialMagneticSampleRate',
1/samplePeriod); 

Do i need to create also the .BIN file? Or is the .csv enough?
Thank you for your time!

xioTechnologies, 2022-05-23:
The .bin file is not used.  I suggest you use the [Python version of this demo](https://github.com/xioTechnologies/Gait-Tracking) because it is easier to understand what is going on from the code.  It also uses the latest version of [Fusion](https://github.com/xioTechnologies/fusion) which is more accurate.