xioTechnologies/Gait-Tracking-With-x-IMU #41: Issue tracking code

celinlab, 2022-04-15:
I try to work on your tracking code but I have this issue : 

>> xIMUdataClass
Index exceeds the number of array elements (0).

Error in xIMUdataClass (line 29)
            obj.FileNamePrefix = varargin{1};

Thank you

xioTechnologies, 2022-04-15:
The code works with the data provided.  If you get an error after changing the data then you will have to do some debugging.  I cannot offer to debug your data.

I suggest you consider using the [Python version](https://github.com/xioTechnologies/Gait-Tracking) of this demonstration as the code may be easier to follow.

closed 2022-04-15 by xioTechnologies

celinlab, 2022-04-15:
Thank you for your response ! 

I try to make work with the data provided but I have this issue. 

We have to fill some data on the matlab code ? To import data in csv, we have to convert them on .mat ?