xioTechnologies/Gait-Tracking-With-x-IMU #44: Real Time System. Speed correction

Flint010, 2022-12-19:
Hello, I am trying to implement this algorithm in real time. Faced the problem of speed correction. How can I calculate speed drift in real time? Or do I need to use a different algorithm or approach?

This implementation will not work in realtime:

velDrift = zeros(size(vel));
stationaryStart = find([0; diff(stationary)] == -1);
stationaryEnd = find([0; diff(stationary)] == 1);
for i = 1:numel(stationaryEnd)
    driftRate = vel(stationaryEnd(i)-1, :) / (stationaryEnd(i) - stationaryStart(i));
    enum = 1:(stationaryEnd(i) - stationaryStart(i));
    drift = [enum'*driftRate(1) enum'*driftRate(2) enum'*driftRate(3)];
    velDrift(stationaryStart(i):stationaryEnd(i)-1, :) = drift;
end

xioTechnologies, 2022-12-19:
A real time implementation must have a latency of one stride to allow for the velocity drift compensation.  I suggest you work with the more recent, [Python implementation](https://github.com/xioTechnologies/Gait-Tracking) as the code is simpler.

Flint010, 2022-12-19:
Thank you so much!