xioTechnologies/Gait-Tracking-With-x-IMU #45: Gait Plot not representative of gait test

psdii, 2023-04-13:
I have data from a different IMU, but programmed to get angular velocity and raw acceleration. I tried using your script, but the results are not reflective of the walking test I did. The plot is showing that I'm not only falling in the Z axis, but walking diagonally, at exaggerated distances. The test I ran only had me walking in a straight line. Would you know what factors could be causing this? My sensor measures at 50 Hz for a 6 meter walk. Is the test too short of a length, or is the frequency bad, or am I using the wrong data? When we put in a specific start and stop time, it does look representative of a walk test, but it is slightly inaccurate.
[straightlinewalktest.csv](https://github.com/xioTechnologies/Gait-Tracking-With-x-IMU/files/11217392/straightlinewalktest.csv)

xioTechnologies, 2023-04-13:
I suggest you use the latest the [latest version](https://github.com/xioTechnologies/Gait-Tracking) and search through closed issues for people who have reported and solved the same problem.

psdii, 2023-04-19:
Previously, my data wasn't formatted so that the stationary periods were not being detected, but now I am able to get a plot that is somewhat indicative of a walk. However, the plot still drifts in the x y and z plane. Would you know of any factors, besides inaccurate sensor data, that would be causing this? Could data frequency be a factor? I am measuring at 50Hz, is the frequency too low that it miscalculates the z x and y distance traveled?

xioTechnologies, 2023-04-19:
Sample rate will effect accuracy.  50 Hz is low but I cannot comment on whether it is the limiting factor of your system's performance.

closed 2023-04-22 by psdii